miguel5612/MQSensorsLib #13: Value of _RLValue

mustaffxx, 2019-09-27:
The function calibrate() use _RLValue. [Looking](https://github.com/miguel5612/MQSensorsLib/blob/master/src/MQUnifiedsensor.h#L38) for this value, I saw that it is set to:
byte _RLValue = 10; //Value in KiloOhms

Why?
In the [datasheet](https://www.sparkfun.com/datasheets/Sensors/MQ-3.pdf) given it says this resistance is 200k Ohms.
Do all MQ sensor modules use 10k?

miguel5612, 2019-09-28:
[image]

Good morning mustaffxx.
Our reason for assuming that it is 10K is because in the sensors we have acquired we have found this value, however after reviewing the datasheet you indicate and seeing the MQ sensor of other manufacturers we have noticed that these values may differ.
We have added two new methods that you can invoke in your program, if the sensor you have is the MQ3, you can use them in the following way:

// To read the current RL value in KOhms
Serial.println(MQ3.getRL());

// To assign a new value to RL in KOhms:
MQ3.setRL(200); //200KOhms for your case

Thank you for contacting us and we will be publishing the new version in the coming days.

miguel5612, 2019-09-30:
New release was published in arduino library manager (1.0.4)

closed 2019-09-30 by miguel5612

mustaffxx, 2019-10-04:
Thanks. My module follows this pattern of components but the components are in different locations, I will try.