miguel5612/MQSensorsLib #16: Something not right about MQ7 sensor readings

ElVasquito, 2019-10-12:
Hello, I am experimenting with an MQ7 CO sensor, and the measurements obtained from the library seem inaccurate.
First, the library wasn't getting any voltage readings, so after inspection of the header file, I noticed the _VOLT_RESOLUTION variable was being defined as type byte, while it should be a float. After changing the type, the voltage readings were correct.
Now, the returned value for CO is a float, while it should be an uint16_t between 20-2000ppm, which is the range that the datasheet shows this sensor can provide. I took a syringe and got an air sample from a car's exhaust tube (with engine on), and after injecting the air it into the sensor, its light turned on, indicating the detection, and the voltage reading went from 0.6 to 3.1V, but the value shown for CO was only 40-45PPM.
LPG, CH4 and Alcohol values were at 9999PPM.
Maybe the constants used for CO calculation are wrong?
I don't have a CO meter so unfortunately I can not calibrate. But value should have been a lot more than 40PPM of CO for the exhaust tube's sample, isn't it?
Thanks for the lib, I will experiment with MQ9 and MQ135 sensors as well in the future ;)

Ghiordy, 2019-10-14:
Hi!
I have checked your troubles, please excuse the time spent to communicate
with you. Our team count with a board where MQ-2 to MQ-9 sensors are
assembled, with it we'll calibration test and verify spans delivered by
factory.
The results and report will be on Notebooks folder, for the next release
with something another fixed issues.
Thanks for you contribution!

ElVasquito, 2019-10-15:
Hi Ghiordy!
Thanks for checking!
I found this Risk Chart in CO levels: https://www.myhomecomfort.org/wp-content/uploads/2015/09/CO_Levels_Risk_Chart.pdf
It mentions that new vehicle emmissions should be 2000ppm. I found OK that other components read 9999PPM, but CO only 40PPM seemed a bit low for me, but maybe I need a bigger air sample, or do the 60-90 cycles instead of having the shield directly connected to 5V. This is the shield I am testing: http://arduinolearning.com/wp-content/uploads/2016/03/mq-7.jpg
Please confirm once you publish the report, thank you for your reply!
Por cierto, podemos hablar en español, o seguimos en inglés?

miguel5612, 2019-10-19:
1. Podrias escrbirnos en español, no hay ningun problema.
2. El sensor MQ7 tiene una region de operacion definida por el datasheet entre 50 y 5000 ppm, lo cual deberia ser un amplio rango para trabajar. Sin embargo no me queda muy claro lo que mensionas en esta parte del texto: (...) but CO only 40PPM (...), estos 40 ppm son resultado de alguna medicion que has realizado, si lo es asi, podrias explicarnos de que forma estas experimentando y por que crees que debe tener un valor mas grande?

miguel5612, 2019-10-24:
I will close this issue because long time with no response, if the problem persist, open new one, thanks ^_^

closed 2019-10-24 by miguel5612

ElVasquito, 2019-10-28:
Hola, perdón la demora en responder. Como decía, no tengo equipamiento para calibrar. Simplemente inyecté el aire del escape del auto en el sensor, e imaginé que con más de 3V de detección la medición sería mayor a 40ppm. Quizás imaginé mal :)
Los otros valores sí subieron mucho, y pueden utilizarse como referencia de todas formas. Gracias.