miguel5612/MQSensorsLib #17: incompatibility with wemos D1 mini

Gazz-yniere, 2019-10-16:
I'm trying to use the library with a wemos D1 mini (esp8266) but I have some issues. 
After compiling I have warnings that said Architecture of the library is for AVR and i'm using ESP8266 and it may not be compatible.
When the code is running on the Wemos I encounter a error of illegal instruction:

11:59:27.692 -> Exception (0):
11:59:27.692 -> epc1=0x4000dce5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
11:59:27.772 -> 
11:59:27.772 -> >>>stack>>>
11:59:27.772 -> 
11:59:27.772 -> ctx: cont
11:59:27.772 -> sp: 3ffffd70 end: 3fffffc0 offset: 01a0
11:59:27.852 -> 3fffff10:  00000000 00000000 00000000 3ffee52c  
11:59:27.892 -> 3fffff20:  00000010 3fffff80 3fffff80 40203f1f  
11:59:27.932 -> 3fffff30:  3ffe85b0 00000002 3fffff80 40203f6b  
11:59:27.972 -> 3fffff40:  cccccccd 40588ccc 3fffff80 40203fb4  
11:59:28.052 -> 3fffff50:  3fffdad0 3ffee480 3fffff80 40203fea  
11:59:28.092 -> 3fffff60:  3ffe85b0 3ffee480 3ffee480 3ffee52c  
11:59:28.132 -> 3fffff70:  3fffdad0 3ffee480 3ffee4fc 4020276a  
11:59:28.172 -> 3fffff80:  3ffef26c 0000000f 00000002 feefeffe  
11:59:28.252 -> 3fffff90:  feefeffe 00000000 3ffee4fc 3ffee52c  
11:59:28.292 -> 3fffffa0:  3fffdad0 00000000 3ffee4fc 40204390  
11:59:28.332 -> 3fffffb0:  feefeffe feefeffe 3ffe8504 40100ce1  
11:59:28.372 -> <<<stack<<<

sounds to be an :

Exception 0: Illegal instruction
PC: 0x4000dce5
EXCVADDR: 0x00000000

I tried on to Wemos with differents sensors (MQ2 MQ135 and MQ7), change setting of compiling from arduinoISP to AVR ISP but nothing change.

I'm missing something? or there is a bug?

chrisi51, 2019-10-16:
Same on my Side ... ESP8266 with MQ135 ... it gave me approx 10 times values for all kinds of gas and after that it starts with the same exception ... now if i reboot it or turn off and power on again it directly starts with that exception.

What i can say is, that Voltage reading seems to be allways 0 ... so probably division by zero
In the first outputs it was getting lower and lower until it started throwing this exception.

20:12:11.465 -> Exception (0):
20:12:11.500 -> epc1=0x4000dce5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
20:12:11.570 -> 
20:12:11.570 -> >>>stack>>>
20:12:11.570 -> 
20:12:11.570 -> ctx: cont
20:12:11.605 -> sp: 3ffffd40 end: 3fffffc0 offset: 01a0
20:12:11.639 -> 3ffffee0:  00000000 00000000 ff000055 00000073  
20:12:11.673 -> 3ffffef0:  00000065 00000074 00000055 3ffee774  
20:12:11.742 -> 3fffff00:  3ffe85da 00000002 3fffff50 40207242  
20:12:11.776 -> 3fffff10:  00000000 00000000 3fffff50 402072a0  
20:12:11.845 -> 3fffff20:  3ffee720 41bf3333 3fffff50 402072e0  
20:12:11.880 -> 3fffff30:  3ffe85da 00000000 3ffee758 3ffee774  
20:12:11.947 -> 3fffff40:  3ffee720 41bf3333 00000002 40201202  
20:12:11.981 -> 3fffff50:  00004f43 00000000 00fee7e8 40206131  
20:12:12.049 -> 3fffff60:  00003d9c 00000001 3ffee7e8 40206131  
20:12:12.084 -> 3fffff70:  3ffe877e 00000001 3ffee7e8 3ffee904  
20:12:12.151 -> 3fffff80:  3fffdad0 00000037 3ffee758 3ffee904  
20:12:12.185 -> 3fffff90:  3fffdad0 00000000 3ffee7e8 40201727  
20:12:12.253 -> 3fffffa0:  feefeffe feefeffe 3ffee8d4 40208394  
20:12:12.288 -> 3fffffb0:  feefeffe feefeffe 3ffe8518 401005a9  
20:12:12.356 -> <<<stack<<<

chrisi51, 2019-10-16:
taking another MQ135 it starts again from showing me 5V which gets lower on every cycle (15sec) until it gets 0V after some cycles.

chrisi51, 2019-10-16:
btw. that is the datasheet: https://cdn.shopify.com/s/files/1/1509/1638/files/MQ-135_Gas_Sensor_Modul_Datenblatt.pdf?6546122777554514074 where RL is defined with "adjustable"

miguel5612, 2019-10-19:
Hi, chrisi51 
Sorry for delay, can you please add your source  code, to test and identify the problem?.
Thanks

Note: This library now is not fully compatible with wemos and another development boards, we recomend for that use join arduino and wemos together using serial protocol and share the measures. Wemos is a good as a network card.

miguel5612, 2019-10-24:
I will close this issue because long time with no response, if the problem persist, open new one, thanks ^_^

closed 2019-10-24 by miguel5612