miguel5612/MQSensorsLib #18: Incorrect MQ7 sensor readings

YiannisBourkelis, 2019-10-24:
I am running the example for the MQ7 sensor, connected on a ESP32 devkit board but I get incorrect results from the library. What might be wrong here?
Bellow is the serial output

Lectures for MQ-7
Volt: 5.00 V
R0: 4.00 Ohm
H2: 9999.00 ppm
LPG: 9999.00 ppm
CH4: 9999.00 ppm
CO: 9999.00 ppm
Alcohol: 9999.00 ppm
***************************

miguel5612, 2019-10-24:
Hi! YiannisBourkelis, thanks for using MQSensorLibrary.
There are some errors to fix:
1. The correct voltage levels in the analog pins are in this range: 0-3v3. for that reason you should use in your setup this code line:
MQ9.setVoltResolution(3.3);
2. Try this: 
* Connect your analog pin to GND and call the function MQ9.getVoltage(true); to know if our library is reading correctly the analog pin. The expected measure in this case, is 0.
* Connect your analog pin to VCC (3.3) and call the function MQ9.getVoltage(true); to know if our library is reading correctly the analog pin. The expected measure in this case, is 3.3.
When step two (2) is complete, you can use normally your sensor and this will work correctly.
I look forward to your comments.

YiannisBourkelis, 2019-10-25:
Thank you @miguel5612 , 
I tried MQ7.setVoltResolution(3.3), but I still get incorrect results. 
Maybe because ESP32 ADC has 12bits resolution (values in range 0-4095) and not 1024 like most arduino.
Here is the output from your example:

***************************
Lectures for MQ-7
Volt: 3.60 V
R0: 4.00 Ohm
H2: 9999.00 ppm
LPG: 9999.00 ppm
CH4: 9999.00 ppm
CO: 9999.00 ppm
Alcohol: 9999.00 ppm
***************************

Example code:
```
/*
  MQUnifiedsensor Library - reading an MQ7

  Demonstrates the use a MQ7
   sensor.
  Library originally added 01 may 2019
  by Miguel A Califa, Yersson Carrillo, Ghiordy Contreras, Mario Rodriguez
 
  Added example
  modified 23 May 2019
  by Miguel Califa 

 This example code is in the public domain.

*/

//Include the library
#include <MQUnifiedsensor.h>

//Definitions
#define pin 39 //Analog input 0 of your arduino
#define type 7 //MQ7
//#define calibration_button 13 //Pin to calibrate your sensor

//Declare Sensor
MQUnifiedsensor MQ7(pin, type);

//Variables
float H2, LPG, CH4, CO, Alcohol;

void setup() {
  Serial.begin(115200); //Init serial port
   //init the sensor
  /*****************************  MQInicializar****************************************
  Input:  pin, type 
  Output:  
  Remarks: This function create the sensor object.
  ************************************************************************************/ 
  MQ7.setVoltResolution(3.3);
    
  MQ7.inicializar(); 
  //pinMode(calibration_button, INPUT);
}

void loop() {
  MQ7.update(); // Update data, the arduino will be read the voltaje in the analog pin
  /* 
    //Si el valor de RL es diferente a 10K por favor asigna tu valor de RL con el siguiente metodo:
    MQ7.setRL(10);
  */
  /*
  //Rutina de calibracion - Uncomment if you need (setup too and header)
  if(calibration_button)
  {
    float R0 = MQ7.calibrate();
    MQ7.setR0(R0);
  }
  */
  /*****************************  MQReadSensor  ****************************************
  Input:   Gas - Serial print flag
  Output:  Value in PPM
  Remarks: This function use readPPM to read the value in PPM the gas in the air.
  ************************************************************************************/ 
  //Read the sensor and print in serial port
  //Lecture will be saved in lecture variable
  //float lecture =  MQ7.readSensor("", true); // Return CH4 concentration
  // Options, uncomment where you need
  H2 =  MQ7.readSensor("H2"); // Return CH4 concentration
  LPG =  MQ7.readSensor("LPG"); // Return LPG concentration
  CH4 =  MQ7.readSensor("CH4"); // Return CH4 concentration
  CO =  MQ7.readSensor("CO"); // Return CO concentration
  Alcohol =  MQ7.readSensor("Alcohol"); // Return Alcohol concentration
  
  Serial.println("***************************");
  Serial.println("Lectures for MQ-7");
  Serial.print("Volt: ");Serial.print(MQ7.getVoltage(true));Serial.println(" V"); 
  Serial.print("R0: ");Serial.print(MQ7.getR0());Serial.println(" Ohm"); 
  Serial.print("H2: ");Serial.print(H2,2);Serial.println(" ppm");
  Serial.print("LPG: ");Serial.print(LPG,2);Serial.println(" ppm");
  Serial.print("CH4: ");Serial.print(CH4,2);Serial.println(" ppm");
  Serial.print("CO: ");Serial.print(CO,2);Serial.println(" ppm");
  Serial.print("Alcohol: ");Serial.print(Alcohol,2);Serial.println(" ppm");
  Serial.println("***************************");  

  delay(1000);
}
```

Dave-Azcarate, 2019-11-01:
Hello I was watching your videos and I want to use the mq7 sensor, so I read a lot of datasheet about it, but in you code It doesn't change the voltage, according to teh datasheet the Mq7 has two phases, the firs is the preheat at 5volts on the phase the CO is consumed and is prepared to the nex read, the heating phase take around 60 seconds, the next phase is at 1.4 volts and takes around 90 seconds, at the end of this phase the voltage combe back to 5 volts and fast it takes the value on the input, and it comes a loop, please check it!!! I will be grateful!!

miguel5612, 2019-12-08:
It is an interesting characteristic, we will analyze it and organize a structure to implement it.

miguel5612, 2020-03-31:
Errors on calibration was fixed on the release 2.0 , please update and it will works well
[image]

closed 2020-03-31 by miguel5612