miguel5612/MQSensorsLib #19: Error in MQ4 sensor calibrating

Dimitrah, 2019-11-14:
at examples/MQ-4/MQ-4.ino there is an error in the if function:
//Rutina de calibracion - Uncomment if you need (setup too and header)
  if(calibration_button)
  {
    float R0 = MQ4.calibrate();
    MQ4.setR0(R0);
  }

calibration_button=13 and so the if statement is always true.
Instead of if(calibration_button) should be if(digitalRead(calibration_button==High)).

miguel5612, 2019-12-08:
Hello Dimitrah.
We are processing your request, in the coming weeks we will be publishing the steps to implement the solution :). We appreciate your patience

miguel5612, 2020-03-31:
This was fixed on 2.0 version, please update and it will work well.
[image]

If problem continues, please open a new issue

closed 2020-03-31 by miguel5612