miguel5612/MQSensorsLib #20: Support digital MQ sensors

filips123, 2019-11-30:
**Is your feature request related to a problem? Please describe.**

Based on examples of this library, this library supports analogue MQ sensors. However, Arduino-like devices usually don't have much analogue input pins. This can be a problem if you want to connect more sensors at once.

**Describe the solution you'd like**

Luckily, most MQ sensors also support digital mode. I don't know if all of them supports it, but I'm sure that at least MQ-2, MQ3, MQ-7 and MQ-135 have `DOUT`/`DO` pin. I also don't know which protocol they use. Based on some datasheets, they use some "TTL signal".

It would be good that this library would also support them.

miguel5612, 2019-12-08:
Good day flips, we regret the delay, we have carried out the analysis and we have determined that the issues you have raised will be resolved in the following order:
Check compatibility with the digital output.
2. Compatibility with NodeMcu
3. Connection with external A/D converters to a nodemcu.
In the following weeks we will be answering your questions one by one, we appreciate your patience.

miguel5612, 2020-03-31:
Digital sensors is interesting feature but is very simple to implement into library, but we added an example showing the instructions and how to implement.
Thank you friend :), good feature 👯‍♂

closed 2020-03-31 by miguel5612