miguel5612/MQSensorsLib #21: Support for NodeMCU

filips123, 2019-11-30:
Does this library supports NodeMCU? If not, can it be updated to support it?

miguel5612, 2019-12-08:
Good day flips, we regret the delay, we have carried out the analysis and we have determined that the issues you have raised will be resolved in the following order:
Check compatibility with the digital output.
2. Compatibility with NodeMcu
3. Connection with external A/D converters to a nodemcu.
In the following weeks we will be answering your questions one by one, we appreciate your patience.

miguel5612, 2020-03-31:
NodeMcu support was added on the release 1.10, please update your library and load your code on the ESP, please take care with voltage managment. Review the readme, any issue contact us :)
[image]

closed 2020-03-31 by miguel5612