miguel5612/MQSensorsLib #22: Support for external A2D convertors

filips123, 2019-12-03:
**Is your feature request related to a problem? Please describe.**

Some microcontrollers, like NodeMCU, only have limited number of analog inputs. This means that you cant use more than one sensor, or even none if controller does not have any analog inputs.

**Describe the solution you'd like**

Solution is to add support for external I2C analog-to-digital convertors. That convertors, which have more analog inputs, can then be attached to controllers. This library should then handle them and read and parse values normally as with internal analog inputs.

As there are quite many A2D convertors, system should be modular and allow more A2D convertors and should be easy to add more.

miguel5612, 2019-12-08:
Good day flips, we regret the delay, we have carried out the analysis and we have determined that the issues you have raised will be resolved in the following order:
Check compatibility with the digital output.
2. Compatibility with NodeMcu
3. Connection with external A/D converters to a nodemcu.
In the following weeks we will be answering your questions one by one, we appreciate your patience.

filips123, 2020-01-23:
So, I checked a few things and I think I know how should external convertors be supported. If this is true, the user can support them even without heavy modifications to this original library.

1. I will assume that the user already has a library for communication with a convertor, like [this](https://github.com/adafruit/Adafruit_ADS1X15) for Adafruit ADS1X15.

2. User overwrites `MQUnifiedsensor::getVoltage` to read from the converter instead of `analogRead`.

3. User uses `MQUnifiedsensor::setVoltResolution` and sets it to the resolution of the convertor.

4. Then the library should read from external convertor instead of the built-in one.

However, I'm not sure how exactly can I overwrite `MQUnifiedsensor::getVoltage` in a way that other parts of the library will use new overwritten version.

Gazz-yniere, 2020-01-29:
Hello,
if you did not find a solution already, I use multiple analog sensor on the same analog input with a CD4051. and after that you do not need to change something in the code. 

Here multiple explanations:
https://github.com/witnessmenow/ESP8266-4051-Multiplexer-Example
https://www.youtube.com/watch?v=S5zp_S3vo1Y
https://www.instructables.com/id/How-to-Use-Multiple-Analog-Sensors-on-Your-ESP8266/

For me It works like a charm and it's very simple to implement.

miguel5612, 2020-03-31:
External A2D option was added on 2.0 release, please update and load the example.
[image]
This presentation has the instruccions.
If you have any problem, open a new issue, that was a good feature :)
https://github.com/miguel5612/MQSensorsLib_Docs/blob/master/Docs/MQSensorLib_2.0.pdf

closed 2020-03-31 by miguel5612