miguel5612/MQSensorsLib #23: only 9999.00 ppm

fatihgvn, 2019-12-31:
**Describe the bug**
code in arduino
```
/*
  MQUnifiedsensor Library - reading an MQ135

  Demonstrates the use a MQ135 sensor.
  Library originally added 01 may 2019
  by Miguel A Califa, Yersson Carrillo, Ghiordy Contreras, Mario Rodriguez
 
  Added example
  modified 23 May 2019
  by Miguel Califa 

 This example code is in the public domain.

*/

//Include the library
#include <MQUnifiedsensor.h>

//Definitions
#define pin A0 //Analog input 0 of your arduino
#define type 135 //MQ135
//#define calibration_button 13 //Pin to calibrate your sensor

//Declare Sensor
MQUnifiedsensor MQ135(pin, type);

//Variables
float CO, Alcohol, CO2, Tolueno, NH4, Acetona;

void setup() {
  Serial.begin(9600); //Init serial port
   //init the sensor
  /*****************************  MQInicializar****************************************
  Input:  pin, type 
  Output:  
  Remarks: This function create the sensor object.
  ************************************************************************************/ 
  MQ135.inicializar(); 
  //pinMode(calibration_button, INPUT);

  MQ135.setVoltResolution(5);
}

void loop() {
  MQ135.update(); // Update data, the arduino will be read the voltaje in the analog pin
  
  /*
    //Si el valor de RL es diferente a 10K por favor asigna tu valor de RL con el siguiente metodo:
    MQ135.setRL(10);
  */
  /*
  //Rutina de calibracion - Uncomment if you need (setup too and header)
  if(calibration_button)
  {
    float R0 = MQ135.calibrate();
    MQ135.setR0(R0):
  }
  */
  
   /*****************************  MQReadSensor  ****************************************
  Input:   Gas - Serial print flag
  Output:  Value in PPM
  Remarks: This function use readPPM to read the value in PPM the gas in the air.
  ************************************************************************************/ 
  //Read the sensor and print in serial port
  //Lecture will be saved in lecture variable
  //float lecture =  MQ135.readSensor("", true); // Return NH4 concentration
  // Options, uncomment where you need
  CO =  MQ135.readSensor("CO"); // Return CO concentration
  Alcohol =  MQ135.readSensor("Alcohol"); // Return Alcohol concentration
  CO2 =  MQ135.readSensor("CO2"); // Return CO2 concentration
  Tolueno =  MQ135.readSensor("Tolueno"); // Return Tolueno concentration
  NH4 =  MQ135.readSensor("NH4"); // Return NH4 concentration
  Acetona =  MQ135.readSensor("Acetona"); // Return Acetona concentration
  
  Serial.println("***************************");
  Serial.print("Volt: ");Serial.print(MQ135.getVoltage(true));Serial.println(" V"); 
  Serial.print("R0: ");Serial.print(MQ135.getR0());Serial.println(" Ohm"); 
  Serial.print("CO: ");Serial.print(CO,2);Serial.println(" ppm");
  Serial.print("Alcohol: ");Serial.print(Alcohol,2);Serial.println(" ppm");
  Serial.print("CO2: ");Serial.print(CO2,2);Serial.println(" ppm");
  Serial.print("Tolueno: ");Serial.print(Tolueno,2);Serial.println(" ppm");
  Serial.print("NH4: ");Serial.print(NH4,2);Serial.println(" ppm");
  Serial.print("Acetona: ");Serial.print(Acetona,2);Serial.println(" ppm");
  Serial.print("Analog: ");Serial.print(analogRead(pin), DEC);Serial.println(" "); // my added
  Serial.println("***************************");
  
  delay(500);
}
```

**Serial Monitor**

A0-A0
```***************************
Volt: 0.14 V
R0: 76.63 Ohm
CO: 9999.00 ppm
Alcohol: 9999.00 ppm
CO2: 9999.00 ppm
Tolueno: 9999.00 ppm
NH4: 9999.00 ppm
Acetona: 9999.00 ppm
Analog: 28 
***************************
```
A0-GND
```***************************
Volt: 0.02 V
R0: 76.63 Ohm
CO: 9999.00 ppm
Alcohol: 9999.00 ppm
CO2: 9999.00 ppm
Tolueno: 9999.00 ppm
NH4: 9999.00 ppm
Acetona: 9999.00 ppm
Analog: 4 
***************************
```
A0-VCC
```***************************
Volt: 4.99 V
R0: 76.63 Ohm
CO: 9999.00 ppm
Alcohol: 9999.00 ppm
CO2: 9999.00 ppm
Tolueno: 9999.00 ppm
NH4: 9999.00 ppm
Acetona: 9999.00 ppm
Analog: 1020 
***************************
```
**Circuit Picture**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 18
 - Ide Version 1.8.10

**Explanation**
Pin connections:
  - A0-A0
  - Gnd-Gnd
  - VCC-5V

I've been trying for a few days. Where am I wrong?
Only 9999.00 ppm data comes from the serial.

iMustaafa, 2019-12-31:
Even i'm facing the same problem

MichaelKostin, 2020-01-07:
The same issue

NarinLab, 2020-03-24:
happy 9999 on all MQ7, MQ131, and MQ135.
This library is promising but after implementing it i just got 9999

miguel5612, 2020-03-26:
For this, i will be implement a new table in the next release, i will share you the demo:
|ADC_In | Ecuacion_V_ADC | Voltaje_ADC |        Ecuacion_R         | Resistencia_RS |    EQ_Ratio  | Ratio (RS/R0) | Ecuacion_PPM |     PPM    |
***********************************************************************************************************************************************

miguel5612, 2020-03-31:
Calibrations issues was fixed on new versions 1.9.9, 1.10, 2.0.
Thank you :).
Please update your library and it will solved! 👯‍♂ 
[image]

closed 2020-03-31 by miguel5612