miguel5612/MQSensorsLib #24: Translate the Spanish strings / words to English

SeppPenner, 2020-03-17:
**Is your feature request related to a problem? Please describe.**
Not a problem, just the examples and some function names (checkout https://github.com/miguel5612/MQSensorsLib/blob/master/examples/MQ-2/MQ-2.ino --> `//Si el valor de RL es diferente a 10K por favor asigna tu valor de RL con el siguiente metodo:` is Spanish or the method `inicializar`.

**Describe the solution you'd like**
A clean library with everything including the examples in English so that everyone can understand it.

**Describe alternatives you've considered**
Alternative for me would be to fork an change the library on my own but I believe it's better to contribute here if possible.

**Additional context**
None.

miguel5612, 2020-03-26:
Excellent, this feature will implement this moth, thank you! :D

miguel5612, 2020-03-31:
Version 1.10 fixed words on spanish, thank you! :)

closed 2020-03-31 by miguel5612

SeppPenner, 2020-03-31:
Nice :)