miguel5612/MQSensorsLib #27: Calibration Methodology (for MQ7)

amirsalman93, 2020-04-11:
If you look at the datasheet of MQ7, it says that R0 is the resistance of sensor in 100 ppm of CO.
[image]

In this library, it assumes R0 to be the resistance of sensor in clean air. Where as, the wording from the datasheet are R0: sensor resistance at 100 ppm CO in the clean air. which means that there is no other gas in the air except 100 ppm of CO.

I believe we need a source of CO (with known concentration in ppm) which we need to use to expose to our sensor. and then measure R0.

What is your opinion?

Thanks

miguel5612, 2020-04-15:
Hi amirsalman93.
Everything you say is correct, calibration should be performed by subjecting the sensor after preheating to an atmosphere that meets the conditions described by the manufacturer, which in this case is 100 ppm of CO. Supported by a previously calibrated instrument that guarantees that it is neither more nor less than 100 ppm.
We, in order to simplify the use of the library and avoid it to give very high values, execute the calibration routine in the setup, but nevertheless we leave a comment for the people that make their use in productive measurement instruments (that is to say, whose data will be used for analysis and other applications, not only for detection) we recommend that the calibration routine is only done when a button is pressed, button that should be pressed only when the sensor has preheated and is in that atmosphere that has been established by the manufacturer.

closed 2020-04-15 by miguel5612

amirsalman93, 2020-04-19:
Sounds Good. Thanks.