miguel5612/MQSensorsLib #28: MQ131 doubt

atlask1, 2020-04-13:
Hi, I have some doubt about MQ131 example and parameters.
First of all, if 'b' is the result of ' log(y) - m * log(x)' , what's 'a' ?
You link a datasheet for MQ131 that consider only an high concentration sensor for ozone, but Ostaquet lib and Winsen datasheet tell about 2 kind of sensor, black bachelite for low concentration and metal case for high concentration.
There's a lot of difference from ppm or ppm ... and the graph of Winsen respect sensorportal is the opposite.
It's also not clear (in datasheets) if is Rs/R0 or R0/Rs because one tell about ratio go up if concentration go up, in the other ratio go down if concentration go up.
thanks

atlask1, 2020-04-13:
Could you also explain me how you calculate a and b where   a = 23,943 and b = -1.11?
Thanks in advance

miguel5612, 2020-04-15:
That's right, here we definitely have to make a change in the library because it is R0/RS for this sensor, thanks for your contribution.
Second, the values of "a" and "b" depend on the mathematical model you use, we have captured several points of the graph that they indicate in the datasheet and supported by excel and python we have elaborated the linear and exponential regressions.
The linear analysis can be found here:
https://github.com/miguel5612/MQSensorsLib_Docs/tree/master/Internal_design_documents
The exponential analysis you find here:
https://github.com/miguel5612/MQSensorsLib_Docs/tree/master/WPDigitalizer
And in each example there is a table in commentary with the exponential regression values and the exponential model is the one we currently recommend because it better models the curve shown in the manufacturer's data sheet.

miguel5612, 2020-04-15:
About the two types of sensors, we are using the one shown in the datasheet, the second one you haven't seen before, you could leave me a comment with the link to the datasheet of the other type of MQ-131?, thanks @atlask1

atlask1, 2020-04-15:
Hi,
thanks for you answer.
this is link for MQ131 Winsen LOW concentration 

[https://www.winsen-sensor.com/d/files/PDF/Semiconductor%20Gas%20Sensor/mq131-(low-concentration)-ver1_3-manual.pdf](url)

and this is link for MQ131 HIGH concentration

[https://www.winsen-sensor.com/d/files/PDF/Semiconductor%20Gas%20Sensor/mq131(high-concentration)-ver1_4-manual.pdf](url)

as you will see look like a 'copy&paste' from other doc, there's  somethings strange, graph are the same, change only ppb/ppm , but concentration go up if ratio go up .
Other strange things is the range: from 10 to 1000 ppb or from 10 to 1000 ppm ... but from 0 to 10? 
or better: from 0 to 5 ppm? after 0.8~1.2 ozone is really toxic, how can I think to measure 10 to 1000 ppm outside of a laboratory? 
As you can see, the graph is really different from your.
I don't know if my sensor is a winsen sensor, but I don't find any other doc.
I have try to write to Winsen but no one answer me.
thanks!

miguel5612, 2020-04-21:
Following this datasheet, used to made the [library](https://www.sensorsportal.com/DOWNLOADS/MQ131.pdf) 
Our range on PPM is: 10-1000ppm Ozone 
Is **high concentration MQ-131 sensor.**
We will make the correction in the mathematical equation for the MQ-131 sensor because it is R0/RS, the opposite of the other sensors.
If you have a low concentration MQ-131 sensor, we recommend to calculate the values of A and B and change them in the example MQ-131 and it should work normally.
PS: In case you calculate it, it is good that you reopen this issue or send us an email to update the library, thank you very much.
The exponential calculations are in:
[Exponential](https://github.com/miguel5612/MQSensorsLib_Docs/tree/master/WPDigitalizer)
And [linear](https://github.com/miguel5612/MQSensorsLib_Docs/tree/master/Internal_design_documents)

closed 2020-04-21 by miguel5612

ryzen3100, 2021-04-07:
I am having trouble setting this up for the low concentration model and getting very high readings at around 400 ppm.
Can you share an example  A and B values for the low concentration (black housing) model, please?