miguel5612/MQSensorsLib #29: Problems with MQ135

MalteHillmann, 2020-05-16:
Hi Miguel,

at first: Thank you for your Lib! :)

I have some problems using my MQ135 sensors: They show very strange ppm values. I want to use this sensor for room air quality detection, so i'm mostly interested in CO2 values in ppm range from 0-1000ppm.
I analyzed the board layout and found out, that my sensor board only has an 1 kΩ resistor for RL. This causes the sensor to be very insensitive so I decided to remove that 1 kΩ SMD resistor and add some linear trimmer with up to 50 kΩ. Now I can change my RL to use higher sensitivity. 

The problem is: The sensor shows values of "0.11" ppm in normal air. But normal air should have aprox. 400 ppm of CO2?

I dont even understand your exponential regressions for the MQ135. Where are these values from? I cant find anything like that in the datasheet. The CO2 curve in the RS/R0 graph seems to be "almost" linear?

Maybe I'm doing things wrong, but I cant find any more helpful information. 
Can you help me? :/

Thank you!

Kind regards
Malte

Raphael909, 2020-05-29:
I am kind a seeing same sort of issue with arduino pro min 3.3V.  This is what my constructor looks like `MQUnifiedsensor MQ135("Arduino Pro Mini", 3.3, 10, pin, "MQ-135");` Also, replaced on board resistor with 20kOhms and have voltage divider for A0.

miguel5612, 2020-05-31:
Good morning Raphael909.
Thank you for your question. And next will be the answer.
First, its correct, 200 PPM is CO2 at atmosphere, but when sensor calibrates it is assumed as "0", therefore you should place or add an offset of 200.  This is a correction that I will make in the examples in the next update of the library. And yes, it's fine to switch to 3.3 volts if your card works at 3.3 voltage level, but as explained in the tutorials I expect your MQ sensor to be connected to 5Volts to guarantee the conditions under which its use is recommended and the voltage is reduced at the input of the ADC with a resistive voltage divider.
To solve the problem you have with the RL I recommend you to add this line during the setup of your arduino program, so the library will update the RL value.
MQ135.setRL(1);

closed 2020-05-31 by miguel5612

miguel5612, 2020-05-31:
I have closed the issue because we have updated the code and analyzed the problem, however if you still experience any other error please let us know :)

watama1299, 2020-12-22:
Hi Miguel,

First I want to say thank you to you and your team for putting together this library.

I'm currently using the latest build of the MQUnifiedSensor Library and I'm still having issues regarding measurements made using the MQ-135 sensor. The measurements made through the Arduino are around 1-10 ppm for CO2. Looking at the solution you provided above, you said that I might have to compensate this reading with the current atmospheric CO2 ppm which is around 411 ppm now. Is that still the case since you said that you have updated the code? Or perhaps I implemented the code wrongly. Please advice.

Best regards,
watama1299

MalteHillmann, 2021-01-08:
>Looking at the solution you provided above, you said that I might have to compensate this reading with the current atmospheric CO2 ppm which is around 411 ppm now. Is that still the case since you said that you have updated the code? Or perhaps I implemented the code wrongly. Please advice.

Hey @watama1299, 

I'm very sure @miguel5612 changed this offset only in the MQ135-ALL example. So you have to add the offset yourself or calibrate the sensor with another already calibrated CO2 sensor.

See this:
https://github.com/miguel5612/MQSensorsLib/blob/874be50106d9cbf5fd02478f73e1dbaa94f6c638/examples/MQ-135-ALL/MQ-135-ALL.ino#L109

TPD95100, 2022-02-06:
Dear all,

I am just Reading thru the sources and comments as I want to measure CO2 as well. Does anybody understand why it is written in the comments that 200 PPM is CO2 at atmosphere and then, one should add 400? I know that fresh air has about 400ppm - where does the 200 come from?

thanks

miguel5612, 2022-03-20:
Best regards,
Yes you are right, that was my mistake, I have corrected and deleted the 200 in comments and added 400 as it should be.

ArathTzec, 2023-01-19:
Hola! buen día. 
Una pregunta bro , ocupé las dos librerías que ves a continuación: 
[image]

Mi pregunta es la siguiente, son los mismos programas?; solo que uno agrega 400 ppm no?

k-str50, 2023-03-18:
@miguel5612 ,
thanks a lot for making this lib. It's very helpful to me.
I want to control my 3D printer.
This is the task:

1. read the data
2. write the results to a database and to a LCD display
3. check for the limit for CO2 and CO
4. if the values exceed the limits 
--> send a switch off comand to a plug controller (HS110) to cut the power

Now I have a question to the values generated by the software: What about the other measurement values like CO, NO4....?
Do I need also to correct it for a propper result in ppm?