miguel5612/MQSensorsLib #30: RatioMQ135CleanAir

Raphael909, 2020-06-27:
Hello Gentlemen, 
I was trying to use MQ135 for CO2 detection and was looking into MQ135 example and was bit confused about `#define RatioMQ135CleanAir 3.6//RS / R0 = 3.6 ppm ` . Though example is for NH4, I was wondering if this value is true for CO2 detection. Also, not sure how this value was achieved. 

2. Does your model takes into account temperature and humidity by any chance or does it somehow manages to negate those factors. 

3. What should be ballpark range for R0 when using 3.3V reference.

Here is my spec:
Arduino Pro Min - 3.3V
ADC resolution - 10
Replaced on board resistor on MQ135 with 20K..

As always Thanks a ton for your effort and work!

suprnrdy, 2020-07-15:
You probably need to scale down the output (voltage divider) and then set the Voltage-range #define. I don't see anything related to temperature/humidity, but the MQ-135 has a chart that says what the Rs/Ro is for those temperatures. I'm guessing you can multiply this number by your RatioCleanAir. I'm hoping there's a way to adjust temperature after initialization, but I don't see one. 

[image]

miguel5612, 2020-07-21:
hi @suprnrdy  
Thanks for your suggestion, i will review the theory and on future updates we will implement this feature, i was searching and i found this paper:
https://www.researchgate.net/publication/328875972_Influence_of_Temperature_and_Humidity_on_the_Output_Resistance_Ratio_of_the_MQ-135_Sensor

miguel5612, 2022-03-20:
This change is a bit complex because this correction factor ratio can vary for each MQ sensor of each available manufacturer, therefore the correction that is made from now on is to allow as a special parameter the correction factor, you can implement a small function that depending on the temperature returns the correction factor and this is applied to your reading as follows:

MQ135.readSensor(false, 1.0);

In this example it will add 1.0 to the Rs/R0 value.

berryPiTech, 2024-03-13:
In case someone encounters the same question #1 by the op, the cleanAirRatio 3.6 is just the Rs/Ro ratio for the curve for air (the black horizontal line) in the log-log plot of (Rs/R0) Vs Concentration. This value 3.6 is valid for all gas types detected by MQ-135.  It corresponds to the ratio of Rs/R0 in the absence of any hazardous gases, hence its name cleanAirRatio.