miguel5612/MQSensorsLib #31: Set Temperature to adjust Rs/Ro temperature dependency

suprnrdy, 2020-07-15:
**Is your feature request related to a problem? Please describe.**
I have my MQ-135 in an environment that changes temperatures throughout the day. There are fluctuations throughout the day as the temperature changes, adding this function should help normalize the levels to not be dependent on the changing temperatures.

**Describe the solution you'd like**
Create a setTemp() function that would adjust the calculations based on the temperature/humidity dependency curve. 

**Describe alternatives you've considered**
I've considered re-initializing the library with the value of RatioMQ135CleanAir multiplied by the appropriate factor.

**Additional context**
[image]

miguel5612, 2020-07-21:
hi @suprnrdy  
Thanks for your suggestion, i will review the theory and on future updates we will implement this feature, i was searching and i found this paper:
https://www.researchgate.net/publication/328875972_Influence_of_Temperature_and_Humidity_on_the_Output_Resistance_Ratio_of_the_MQ-135_Sensor

viotemp1, 2021-12-15:
any chance to get this implemented?

miguel5612, 2022-01-07:
Hi sorry for delay, i was busy, could you write me an email, and we stablish how to implement?

miguel5612, 2022-03-20:
This change is a bit complex because this correction factor ratio can vary for each MQ sensor of each available manufacturer, therefore the correction that is made from now on is to allow as a special parameter the correction factor, you can implement a small function that depending on the temperature returns the correction factor and this is applied to your reading as follows:

MQ135.readSensor(false, 1.0);

In this example it will add 1.0 to the Rs/R0 value.

closed 2022-03-20 by miguel5612

viotemp1, 2022-03-20:
Hello, thanks for answer.
I did it like this
    float getCorrectionFactor(float t, float h) {
        if (t < 20){
             return CORA * t * t - CORB * t + CORC - (h-33.)*CORD;
        }
        else {
             return CORE * t + CORF * h + CORG;
        }
    }
following [https://github.com/jehy/arduino-esp8266-mq135/blob/master/mq135-data.h](https://github.com/jehy/arduino-esp8266-mq135/blob/master/mq135-data.h)

Not sure how well will apply to other sensor types...
Regards

miguel5612, 2022-03-20:
Super, thank you very much for your comment.
Since it does not apply to all sensors I have temporarily added the example MQ-135-CorrectionFactor incorporating the code you have attached here.