miguel5612/MQSensorsLib #41: MQUnifiedsensor integration with ESPHome

gsgameiro, 2020-12-20:
I'm not able to make MQUnifiedsensor working with ESPHome. I'm using a tutorial based in *[Custom Sensor Component](https://esphome.io/components/sensor/custom.html?)* but I believe because my code is in C (not C++) I´m not able to declare and initialize in other to set up parameter (board, pin, etc).

When declaring sensor
   MQUnifiedsensor bmp("ESP8266", 3.3, 10, A0, "MQ-135"); //this in the line that doesn't compile correctly
   MQUnifiedsensor bmp; //while with this line compile is success but returns are wrong because of set up not done

[My code .h](https://pastebin.pl/view/87d56597)
[ESPHome compiling errors](https://pastebin.pl/view/ca392e38)
[ESPHome config .yaml](https://pastebin.pl/view/d05e1f30)

**Alternative**
In case there is no way to declare and initialize it, I would like to help you to create methods to initialize (to set up the Board, Pin, Type, Voltage_Resolution, ADC_Bit_Resolution and RatioMQ9CleanAir) if you would like to add it to the library to improve compatibility (I'm new here but I want to help as possible).

I truly believe if we go forward it will get this library more support from the ESPHome community and it will be widely popular on the ESPHome platform.

nukleuz80, 2021-01-05:
Hey there,

i want include it too in esphome.
Is it possible?

nukleuz80, 2021-01-22:
Same to me - i want to include but only errors... Do you have a solution for that?

miguel5612, 2022-03-20:
Currently the library is not configured to be compatible with C and yaml for esphome, however if you detail it the library contains all the values implicit in the examples, you have to implement in ESPHome a function that raises values following the exponential equation and after knowing the voltage, calculate the Rs/R0 value and with the equation obtain the PPM, it would be a reengineering applied to esphome, I would suggest you to create a new repository for it and try it since in this repository you can't implement it because it would affect the compilation for arduino.

Show more on this repo:
https://github.com/miguel5612/MQSensorsLib_Docs

closed 2022-03-20 by miguel5612