miguel5612/MQSensorsLib #43: correction formula for _sensor_volt

GrofuFlorin, 2021-01-15:
although the error is very small especially at a large number of bits the correct formula is
this-> _sensor_volt = (value) * _VOLT_RESOLUTION / (pow(2, _ADC_Bit_Resolution)); 
not 
 this-> _sensor_volt = (value) * _VOLT_RESOLUTION / ((pow(2, _ADC_Bit_Resolution)) - 1); 
because there are pow(2, _ADC_Bit_Resolution) quant 0, 1, ......((pow(2, _ADC_Bit_Resolution)) - 1)

JerryLoveCat, 2021-09-13:
bro. you are wrong. as 10 bit ADC. Max is 1023 instead of 1024.

miguel5612, 2022-03-20:
Change dismissed considering that when you have a resolution of 10 bits the resolution value is 1024 possible values, however if we count the state "0" which is the initial state the range goes from 0 to 1023, therefore the equation is correct.

closed 2022-03-20 by miguel5612