miguel5612/MQSensorsLib #52: MQ-7 sensor connecting with nodeMcu V3

oymmusyri, 2022-04-26:
Hi, I'm using the MQ-7 sensor, and I don't know how to connect it with NodeMcu V3. Is it the same as the following circuit with Arduino?  
If not, please help me connect it the right way and update the code if needed.

[image]

Thanks,

oymmusyri, 2022-04-26:
I see the following circuit, but it is without MOSFET(IRLZ34N): 

[image]

2- the other question is why it stops reading from the sensor after 2.3 minutes. Is that right, or is something wrong in the code?
The datasheet says that this operation is repetitive.

miguel5612, 2022-05-02:
Best regards,
Regarding your question:
1. It is correct, guide yourself as it is for arduino, just be careful with the pin that in the example is the #5:

#define PWMPin 5 // Pin connected to mosfet

2. Remember that about the times, it is indicated that the measurement can be continuous but only if:
2.1 Preheat has been exceeded (48 hours).
2.2 Calibration has been performed - clean air 100 PPM.
2.3 The cycle VH 5 volts, VH 1.4 Volts 60 / 90 seconds is maintained.

So if you wanted to make measurements at smaller intervals you would have to modify the program so that using interrupts or more advanced calculations the arduino maintains the pulse on the mosfet to generate the square wave and in parallel is making measurements,
for which I would recommend for example:
https://www.youtube.com/watch?v=Ol6x5mjeu4w

Remember that if you perform the measurement as above in the video you can collaborate with the library by uploading your code with git or here in the same issue.

Also keep in mind that the voltage levels should be 5 volts and 1.4 volts with a margin of error of plus or minus 0.1, measure them with a multimeter and validate that they are met so that your results are reliable.