miguel5612/MQSensorsLib #55: PPM of MQ135 at clean air

PsySc0rpi0n, 2022-09-06:
Hi.

Why you guys considered RS / R0 = 3.6 for MQ135 sensor?? Datasheet says constant value of 4.4.
Check the following line, please:

https://github.com/miguel5612/MQSensorsLib/blob/27465dfb992595bb5b6bc9c20659333a1bcf4758/examples/Calibration/Calibration.ino#L38

kodakjerec, 2022-12-07:
The author is right. The following png is MQ135 datasheet, "AIR" almost equals 3.6.
I wonder know why not 3.5 or 3.7?
[https://components101.com/sites/default/files/inline-images/MQ135-Sensitvity-Characteri.png](url)

miguel5612, 2023-05-16:
Hello,

Thanks for bringing this up. The MQ135 sensor ratio of RS/R0 in clean air can indeed vary based on the specific environment, sensor production batch, and many other factors. It is generally recommended to calibrate the sensor in your specific environment to get the most accurate readings.

The value of 3.6 is a commonly used approximation but it's not always accurate for every use case. The datasheet of another autors indeed suggests different values, but this is under specific conditions and it might not be accurate for all environments.

If you find that a value gives you more accurate readings in your specific use case, please feel free to use it. The library is designed to be flexible and allows you to set the clean air ratio for calibration as per your requirements.

Thanks for your valuable feedback!

closed 2023-05-16 by miguel5612