miguel5612/MQSensorsLib #62: ESP-32 S2 Mini Compatibility

dani-elnino, 2023-04-08:
I am writing because the documentation states that it supports ESP-32 boards. I am attempting to use the Wemos ESP-32 S2 Mini but I am not able to get it to work. I am unsure of what to put for the board in the Hardware related macros so that it recognizes the correct board and what pins I am able to use. Any feedback would be very much appreciated.

dani-elnino, 2023-04-08:
The board can also be found as Lolin S2 Mini

miguel5612, 2023-05-21:
Hello,

Your issue seems to be related to board definitions and pin assignments, which may vary from board to board. The Wemos ESP-32 S2 Mini (also known as Lolin S2 Mini) should indeed be compatible with the library if the ESP-32 boards are supported.

In general, board definitions are used by Arduino IDE to identify and correctly compile code for different hardware. In the case of your board, the correct definition should be included in the Arduino IDE once you've installed the ESP32 board library. This will usually take the form of a descriptive name (for instance, ESP32 Dev Module for a generic ESP32 board), and you should select the closest matching or generic option.

Now, concerning the specific pins to use, you need to refer to the pinout of the Wemos ESP-32 S2 Mini. Usually, the sensor pins can be connected to any GPIO pin that supports the required communication protocol (analog, digital, I2C, SPI, etc.). You'll have to make sure that the pins you choose in your code match with the physical connections you've made between the board and the sensor.

To help further, I would need more specifics about your wiring and the exact code you're trying to run, but here's a generalized suggestion:

#define BOARD "Wemos ESP-32 S2 Mini"
#define SENSOR_PIN x  //replace x with your actual analog/digital pin

Note that these are just placeholders and should be replaced with the proper values for your board and your sensor. The exact macro names (like BOARD or SENSOR_PIN) may also differ and should be replaced with what your library uses.

If you can't find the correct board definition in the Arduino IDE, you might need to manually add support for your board. This would usually involve adding a URL to the 'Additional Boards Manager URLs' field in the Arduino IDE's preferences and then installing the board support package from the Boards Manager.

I hope this information helps! Let me know if you need more assistance.

closed 2023-09-28 by miguel5612