miguel5612/MQSensorsLib #66: Include coeficient tables to library

Nic30, 2023-07-16:
I would like to avoid duplication of constant in my project. What I did is to copy A,B interpolation coefficients from examples. 

There are 2 things which can be done. 1st is to include some array which will contain coefficients or there could be class for every gas sensor with a proper method. Second variant seems to me as better because it is faster and intuitive.
```cpp
class MQ2GasSensor: public MQGasSensor {
public:
  static const float CLEAN_AIR_PPM = 9.83; // RS / R0
  using MQGasSensor::MQGasSensor;
  
  float getPpmH2() {
     return getInterpolation(987.99, -2.162);
  }
  float getPpmLPG() {
     return getInterpolation(574.25, -2.222);
  }
// ...
}
```

miguel5612, 2023-10-04:
it could be like:

`class MQ2GasSensor : public MQUnifiedsensor {
public:
    MQ2GasSensor(const char* board, float voltageResolution, int bitResolution, int pin, const char* type)
        : MQUnifiedsensor(board, voltageResolution, bitResolution, pin, type) {}
    
    float getPpmH2() {
        setA(987.99);
        setB(-2.162);
        return readSensor();
    }

    float getPpmLPG() {
        setA(574.25);
        setB(-2.222);
        return readSensor();
    }

    float getPpmCO() {
        setA(36974);
        setB(-3.109);
        return readSensor();
    }

    float getPpmAlcohol() {
        setA(3616.1);
        setB(-2.675);
        return readSensor();
    }

    float getPpmPropane() {
        setA(658.71);
        setB(-2.168);
        return readSensor();
    }
};`

Could you reply to all examples, and send us the pull or merge request? @Nic30 
Thanks ;)

miguel5612, 2024-09-05:
Closed due to the absence of an associated pull request for an extended period.

closed 2024-09-05 by miguel5612