miguel5612/MQSensorsLib #67: Sensor won't finish the Calibration process if done in clean air.

LloydAlbin, 2023-09-15:
**Describe the bug**
When using the sample code for ESP32 and updating it for the MQ135 per the MQ135-all sample code, I run into the "Warning: Conection issue, R0 is infinite (Open circuit detected) please check your wiring and supply" unless I use a cigarette lighter's gas in front of the sensor during the calibration process.

**Code**
```
#include "Arduino.h"
#include <WiFi.h>

const char* ssid = ""; // Wifi SSID
const char* password = ""; // Wifi Password

//Include the library
#include <MQUnifiedsensor.h>
/************************Hardware Related Macros************************************/
#define         Board                   ("ESP-32") // Wemos ESP-32 or other board, whatever have ESP32 core.

//https://www.amazon.com/HiLetgo-ESP-WROOM-32-Development-Microcontroller-Integrated/dp/B0718T232Z (Although Amazon shows ESP-WROOM-32 ESP32 ESP-32S, the board is the ESP-WROOM-32D)
#define         Pin                     (32) //check the esp32-wroom-32d.jpg image on ESP32 folder 

/***********************Software Related Macros************************************/
#define         Type                    ("MQ-135") //MQ2 or other MQ Sensor, if change this verify your a and b values.
#define         Voltage_Resolution      (3.3) // 3V3 <- IMPORTANT. Source: https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/
#define         ADC_Bit_Resolution      (12) // ESP-32 bit resolution. Source: https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/
#define         RatioMQ135CleanAir      (3.6) //RS / R0 = 9.83 ppm
/*****************************Globals***********************************************/
MQUnifiedsensor MQ135(Board, Voltage_Resolution, ADC_Bit_Resolution, Pin, Type);
/*****************************Globals***********************************************/

void setup()
{

  //Init the serial port communication - to debug the library
  Serial.begin(9600); //Init serial port
  delay(10);

  //Set math model to calculate the PPM concentration and the value of constants
  MQ135.setRegressionMethod(1); //_PPM =  a*ratio^b

  /*****************************  MQ Init ********************************************/ 
  //Remarks: Configure the pin of arduino as input.
  /************************************************************************************/ 
  MQ135.init(); 
  /* 
    //If the RL value is different from 10K please assign your RL value with the following method:
    MQ135.setRL(10);
  */
  /*****************************  MQ CAlibration ********************************************/ 
  // Explanation: 
  // In this routine the sensor will measure the resistance of the sensor supposedly before being pre-heated
  // and on clean air (Calibration conditions), setting up R0 value.
  // We recomend executing this routine only on setup in laboratory conditions.
  // This routine does not need to be executed on each restart, you can load your R0 value from eeprom.
  // Acknowledgements: https://jayconsystems.com/blog/understanding-a-gas-sensor
  Serial.print("Calibrating please wait.");
  float calcR0 = 0;
  for(int i = 1; i<=10; i ++)
  {
    MQ135.update(); // Update data, the arduino will read the voltage from the analog pin
    calcR0 += MQ135.calibrate(RatioMQ135CleanAir);
    Serial.print(".");
  }
  MQ135.setR0(calcR0/10);
  Serial.println("  done!.");
  
  if(isinf(calcR0)) {Serial.println("Warning: Conection issue, R0 is infinite (Open circuit detected) please check your wiring and supply"); while(1);}
  if(calcR0 == 0){Serial.println("Warning: Conection issue found, R0 is zero (Analog pin shorts to ground) please check your wiring and supply"); while(1);}
  /*****************************  MQ CAlibration ********************************************/ 
  Serial.println("** Values from MQ-135 ****");
  Serial.println("|    CO   |  Alcohol |   CO2  |  Toluen  |  NH4  |  Aceton  |");  
}
[... 102 lines cut ...]
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
|   0.00   |   0.00   |   400.00   |   0.00   |   0.00   |   0.00   |
```

**Expected behavior**
I would expect this app to be able to calibrate properly in clean air. I have also tried putting in a delay(20000); after the MQ135.init(); to see if it just needed 20 seconds to warm up before the calibration process started. This did not help so I also tried 60,000 and it also did not help, _sensor_volt = 0.00 after either 20,000 and 60,000 wait.

**Hardware:**
 - Webmos ESP32 D1 Mini
 - MQ135 Sensor
 - I have tried the following 3 different resistor setups
   - 1K Resistor - from [https://blog.asksensors.com/air-quality-sensor-mq135-cloud-mqtt/](https://blog.asksensors.com/air-quality-sensor-mq135-cloud-mqtt/)
   - 1K & 470 Resistors - from [https://www.hackster.io/thingsboard/part-1-how-to-connect-esp-wroom-32-mq135-and-thingsboard-4a4bb4](https://www.hackster.io/thingsboard/part-1-how-to-connect-esp-wroom-32-mq135-and-thingsboard-4a4bb4)
   - 1.7K & 3.3K Resistors - from Page 9 of [https://github.com/miguel5612/MQSensorsLib_Docs/blob/master/Docs/MQSensorLib_2.0.pdf](https://github.com/miguel5612/MQSensorsLib_Docs/blob/master/Docs/MQSensorLib_2.0.pdf)

miguel5612, 2024-09-05:
My friend, in these cases, we usually check the sensor manually, with a program, such as the example given by Jaycon Systems:

https://www.jaycon.com/understanding-a-gas-sensor/

Once all the setup works correctly and we understand each mathematical equation, we proceed to make the adjustment in the library (if required) and we load an example (if it is for a different board than the ones already saved) or update the example (if it is an existing board) in the library.

Finally with a pull request I can support you to make this setting available to the whole community.

I remain attentive to any new developments

miguel5612, 2025-06-05:
After reviewing the report it was concluded that the failure does not correspond to a library bug, but to an incorrect connection or insufficient pre-heating time. It is suggested to verify the wiring and test the sensor with a simple program before using the library. No code changes are planned for now, although clearer diagnostics may be implemented in the future.