miguel5612/MQSensorsLib #69: Probably Incorrect Readings

dirceu-jr, 2023-10-19:
I'm using _MQUnifiedsensor_ library much like `examples/MQ-135-ALL/MQ-135-ALL.ino` shows.

```
MQ135.setA(605.18); MQ135.setB(-3.937); // Configure the equation to calculate CO concentration value
float CO = MQ135.readSensor();

MQ135.setA(77.255); MQ135.setB(-3.18); //Configure the equation to calculate Alcohol concentration value
float Alcohol = MQ135.readSensor();

// ...
```

But when I get near some "element" to test, like Alcohol or Aceton all the readings changes in a correlated way. See:
MQ135: https://thingspeak.com/channels/2295455/
MQ3: https://thingspeak.com/channels/2295454/

The expected behavior is that when I put some Alcohol near the sensor, only Alcohol has a change in readings.

CHE77, 2023-11-03:
Also, I could not get correct values for CO2.  For any Regression Method i get very small values e.g. 1...3
Adding 400ppm correction does not help. i get 401...403 even when i blow on the sensor.

berryPiTech, 2024-03-14:
HI dirceu-jr,
An MQ sensor is designed to detect one type of gas at a time. It is not designed to detect all the gases at the same time.
When alcohol is present, you have to use the appropriate regression coefficients (a,b) for alcohol. The reading for other gases are  meaningless.

berryPiTech, 2024-03-14:
Hi CHE77,

I noticed the same behavior as you did for my MQ3 sensor testing for alcohol. Very small variation in ppm reading.

I found a solution. When I examined the sensor datasheet, the cleanAirRatio is 1, while the sketch uses cleanAirRatio = 60.  Setting the cleanAirRatio = 1 in the program gives me correct readings.
 
You might want to check your sensor datasheet and see if your cleanAirRatio does indeed have the same value used in the program.

Here is my datasheet:
https://cdn.sparkfun.com/datasheets/Sensors/Biometric/MQ-3%20ver1.3%20-%20Manual.pdf

The cleanAirRatio is the red horizontal line in figure 3.

miguel5612, 2024-09-05:
Closed due to clarification and resolution. As mentioned by @berryPiTech, MQ sensors are designed to detect one type of gas at a time. It's essential to use the appropriate regression coefficients (a, b) for the specific gas being measured, as readings for other gases will not be reliable. Additionally, adjusting the cleanAirRatio to match the sensor's datasheet, as highlighted by @berryPiTech, resolves the issue of incorrect values. Thank you all for your input

closed 2024-09-05 by miguel5612