miguel5612/MQSensorsLib #73: Is RL correct for the 3.3V examples?

bvernham, 2023-12-05:
**Describe the bug**
A clear and concise description of what the bug is.
Not so much a bug but a question
[image]
In this of the ESP8266 example are you considering RL = 5K?  Because the example code is still defaulting to 10K?  If the MQ sensor already had a 10K as RL then by adding the voltage divider to "AO" (which is in parallel to "RL" of 10K) aren't you making RL effectively ~3.33K?
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

berryPiTech, 2024-03-14:
Bvernham raises a valid point.  To avoid reducing the effective RL, the second voltage divider (for level shifting down from 5v to 3.3v) should employ high impedance.

A 100k/200k combination would be better than the (1.7k/3.3k) combination in the diagram.

I have verified the 100k/200k combination works fine for esp32c3.

miguel5612, 2024-09-05:
Closed due to clarification. As noted by @berryPiTech, the concern about reducing the effective RL when adding a voltage divider is valid. To avoid this issue, a higher impedance voltage divider (such as 100k/200k) is recommended, rather than the (1.7k/3.3k) combination shown in the example. This approach has been tested and works well with ESP32C3. Thank you for raising this point.

closed 2024-09-05 by miguel5612