miguel5612/MQSensorsLib #74: possible error in the calculation formula for _RS_Calc = ((_VOLT_RESOLUTION*_RL)/_sensor_volt)-_RL;

berryPiTech, 2024-03-13:
I think there is an error in the library source code (https://github.com/miguel5612/MQSensorsLib/blob/master/src/MQUnifiedsensor.cpp):

The variable _VOLT_RESOLUTION in the equation below should be 5V, not  _VOLT_RESOLUTION, which is 3.3volts in case of esp32.
 _RS_Calc = ((_VOLT_RESOLUTION*_RL)/_sensor_volt)-_RL; //Get value of RS in a gas

The above equation (line 148) is supposed to be the implementation of this equation in this link (https://www.jaycon.com/understanding-a-gas-sensor/).  

RS = [(VC x RL) / VRL] – RL
 
The variables are in figure 6 of the link above.

Specifically Vcc=5V is the vcc to drive the heater of the sensor.  It can't be 3.3volts, which is too low to drive the heater.

The correct formula should be:
 _RS_Calc = ((_VCC*_RL)/_sensor_volt)-_RL;

where VCC = 5volts.

daguerpedro, 2024-06-06:
You're right!
Also, i guess the method 'calibrate' is also incorrect, do you agree?

It should calculate a R0.
R0 is RS in fresh air, according to the sensor manufacturer. 

But this method is getting RS in fresh air and dividing by a predefined RS in clean air. In others words, this method gets a RS/R0 instead of R0.

The point of calibration R0 is to get a RS in fresh air.

Do you agree?

miguel5612, 2024-09-05:
@berryPiTech Thank you for raising these important points about the formula and the calibration method. To address this issue and improve the library, we would appreciate a pull request that introduces a new method to calculate RS using the correct formula
Additionally, it would be helpful if this pull request could include examples demonstrating how to use this method, especially with different sensors and scenarios, such as using a 5V VCC for proper sensor operation. This will ensure users can apply the formula correctly based on their specific setups.
We encourage you to contribute the necessary changes, and your input will be highly valued by the community!

miguel5612, 2025-06-05:
The library now computes the sensor resistance with _VCC rather than _VOLT_RESOLUTION, and exposes setVCC() so the supply voltage can be specified. The calibrate function returns R0 = RS_air/ratioInCleanAir, matching datasheet recommendations. Remember to set _VCC to your sensor’s actual supply (e.g., 5 V) if it differs from the ADC reference. Other than manually setting _VCC when required, the fix appears complete.

closed 2025-06-05 by miguel5612