miguel5612/MQSensorsLib #75: MQ-3 sensor and CH4 gas reading 'ovf' failure

AndroidCrypto, 2024-07-29:
**Describe the bug**
I'm trying to read the CH4 gas concentration with a MQ-3 sensor using these values for 'setA' and 'setB':

```
MQ3.setA(2*10^31);
MQ3.setB(19.01);
float ch4 = MQ3.readSensor();
```
Unfortunately, the library gives an '**ovf**' as output. All other gas readings are working.

**Additional context**
I tested the library on with Arduino Uno R4 Wi-Fi microcontroller and 5V for VCC. The sensor is connected to A0 on the Arduino.

miguel5612, 2024-09-05:
hank you for reporting this issue. It seems that the overflow might be caused by the extremely large value used for setA (2*10^31), which is likely exceeding the limits of the variable type. To resolve this, we suggest adjusting the variable types or using alternative mathematical operations that handle large numbers more efficiently.

If you're able to make these changes and test them, it would be fantastic if you could contribute a merge request with the modifications. This would be a great enhancement to the library, and we appreciate your help in improving it!

miguel5612, 2025-06-04:
Fixed on verificar-resolución-del-error--ovf--en-sensor-mq-3#

closed 2025-06-04 by miguel5612